hobbymachinist/els #1: G-code parser

ebo, 2022-01-19:
Decades ago I wrote a g-code parser using Spirit+ (part of the Boost libraries).  It came complete with runtime overloading of commands (so you could run different flavors of RD274*).  That project was WAY overkill and probably would not fit into what you have in mind, but I could probably salvage chunks to be useful.  So, tell me what you want the parser to do, and maybe PM me, and we can discuss seeing if any of it would be useful.

hobbymachinist, 2022-01-19:
Thanks @ebo, the rough plan is to get a modified version of Grbl integrated as a module to allow CNC control of X and Z axis along with spindle sync. If that works, I can get rid of the bespoke stepper controller in els and rewrite almost all the stock turning operations as canned cycles that can be fed into Grbl.

qtron, 2022-06-02:
Hi HM,
help! i'm stuck.. not familiar w github at all, lost a LOT of time trying to find the documentation, circuit diagram & gerber files etc.. i saw a gpio list but not sure what that all means!
i have built a Russian 2 axis arduino Mega ELS & seems to work, but i much prefer yours with the magic, albeit expensive!! keypad.
- I also can't find info on how U re labelled it, so nicely!!
- Do U have a list of the specs of the ELS (i dont mean characteristics)?
- Can U move the els in 0.01mm increments?
- What is the max allowable backlash your s/w can cope with?
- i saw the vid of the failed reverse taper.. what was wrong? some words would help! 

Without a circuit diagram i have no way of knowing how to set it up, & install/ program the Eprom i saw mentioned.. where did U get it from? How did U program it?

I see its been a while since u were here last, so hope ur ok.

I did download the zip file for the code but cant see how to download the documentation. 
i have no idea what a fork is, pull requests, etc.. does pull mean remove ur files?!!
Sorry but i find github strange indeed.
eg, Why does one need to see the inner workings of a gerber file here?
i have kicad & use JLCPCB quite a bit , could see gerber files but they are all individual - how do U download all at once? surely they are zipped?
Thanks!
Qtron

hobbymachinist, 2022-06-03:
The schematics are split into multiple modules, you can open them up in Kicad to have a look. 

Main board: https://github.com/hobbymachinist/els/blob/main/hw/els/els.sch  or https://github.com/hobbymachinist/els/blob/main/hw/els/els.pdf
LCD breakout https://github.com/hobbymachinist/els/blob/main/hw/lcd/lcd.sch 

> i saw a gpio list but not sure what that all means!

Here's a very brief summary how the pins are mapped, 
https://github.com/hobbymachinist/els/blob/main/src/gpio.h#L25-L81

You are expected to have some knowledge of electronics, microcontrollers and writing firmware. This project is not designed to be consumer ready if you're not able to understand how all of these go together unfortunately.

> Do U have a list of the specs of the ELS 

What do you mean by specs ?

> Can U move the els in 0.01mm increments?

This depends on your servo / stepper driver settings (usually called micro steps), the smallest increment I can move my axes is 0.0025mm

> What is the max allowable backlash your s/w can cope with?

Any arbitrary amount, but this needs to be constant and cannot vary by direction. I have since stopped relying on this as the backlash can drift over time leading to inaccurate movements.  

I would recommend switching over to ball screw if you want 2 axes precision control or use an optical DRO scale and connect that to the ELS. The ELS now supports limited closed loop movement on both axis by using the DRO for feedback and compensates for backlash accordingly.

> i saw the vid of the failed reverse taper.. what was wrong? some words would help!

It was a control loop bug and was not setting the stepper direction pulse correctly, I've fixed it.

> does ur ELS do the calcs on the fly, like Clough42's ELS? 

yes, the micro controller used is very capable. STM32F446RE can handle a lot more if the internal features are use correctly. I'm not taking advantage of all the features yet and it seems to work fine for me.

> Why does one need to see the inner workings of a gerber file here?
> i have kicad & use JLCPCB quite a bit , could see gerber files but they are all individual - how do U download all at once?

If you have Kicad, you can open the project files under the hw directory and have a look, export the gerbers as needed for your purpose but as I said, this project is not going to be ready to use out of the box unless you have some basic knowledge of microcontrollers, electronics and firmware programming.

closed 2022-06-03 by hobbymachinist

hobbymachinist, 2022-06-03:
@qtron if you're not comfortable with a full DIY solution, I would recommend you take a look at Rocketronics ELS https://www.rocketronics.de/en/els/. It is ready to use and comes with excellent documentation.

qtron, 2022-06-03:
hiya HM,
i am comfortable! as i said i built the russian 2 axis els..(ver 7e2)
its not so easy to use.

i just need the info. of your nice creation to proceed, if possible.
Yours is the only one with knurling,
  (as well as ricketronics does have issues.
a guy on youtube evaluated it, and some glitches can be seen, like not starting threading! its expensive too ).

pls advise:
does yours do calcs on the fly?

how do i find the cct diagram to load into kicad?
As i said, navigating thru Github trying to find downloads is hard! (obviously not for some!)
PLEASE advise..
thankyou😃

..
Sent from mob. Ph.

qtron, 2022-06-03:
sorry HM, i just found your previous email! much appreciated
the rocketronics unit basic ver 4, delivered is $422 AUD.

***@***.*** if you're not comfortable with a full DIY solution, I would recommend you take a look at Rocketronics ELS https://www.rocketronics.de/en/els/. It is ready to use and comes with excellent documentation.
..
Sent from mob. Ph.

qtron, 2022-06-03:
i am most grateful for the links to the files, but is there a place in github where i can download them, to get into kicad?

by move 0.01mm , i meant move from the control panel, not microstep movement..

Specs: like max settable X & Z motion,
respose time after go button pressed,
lag time of tool vs steps issued,
any step misses at high speed,

absolutely wonderful re closed loop.
i saw the glass scale on your Z motion, (but didnt know if it was just DRO).

BTW,
i saw in MEW that Joe Noci (of Nairobi) uses a rattle damper on his stepper to kill any resonance! it is supposed to help greatly , accel. from zero.

kind regards,

..
Sent from mob. Ph.

hobbymachinist, 2022-06-03:
Go to the repository page, https://github.com/hobbymachinist/els and download the zip file as shown

[image]

> by move 0.01mm , i meant move from the control panel, not microstep movement..

0.005mm

> Specs: like max settable X & Z motion,

There are no soft limits, only what you set the limits to in the program.

> respose time after go button pressed,

Probably 50-100ms. If you're thinking about threading it synchronises to the encoder counter so even if you press it will only start at the spindle position when encoder counter is at a certain value.

> lag time of tool vs steps issued,

speed of light :)   it's probably a few microseconds due to the stepper / servo drivers.

> any step misses at high speed,

not that I have noticed, I use closed loop hybrid steppers that will stall if they cannot keep up but there is no acceleration support for threading - so your mileage may vary depending on your hardware choice.

qtron, 2022-06-03:
thanks again😃
i had already downloaded the zipped MCU f/w.
i mean download the schematics, the wiring hookup diagram to the eprom, & the overall layout..

i saw the list from the links u sent earlier, clicked on ''copy path'', tried 
''permlink copy'' etc. .. nothing happened.
just dunno how to get the sch into kicad,
and how to get the zipped gerber files.
cheers,

..
Sent from mob. Ph.