hobbymachinist/els #4: Questions

kzali, 2022-07-23:
Hi
Firstly I'd like to thank you for such a well though project. I am happy to say that I have gotten all the PCB's and the Neucleo board
The LCD, Encoder, Keyboard, and the hand wheel are on the way. I intend to build it slowly as have beginner level skills (largely in simple Arduino projects.)

I am having a hard time sourcing the FT24C64A EEPROM, CH340C USB UART and the GL850G USB HUB Controller, here in Australia. I understand from your very detailed technical
documentation that I can do without the CH340C and the GL850G (and their passive components) I will however directly wire PA9 and PA10 to the CN3 connectors marked RXand TX on the Nucleo.

There seem to be many i2c 8 pin 64k EEPROMS on au.rsonline and other suppliers! Any suggestions/alternate will be greatly appreciated.

I am not clear what this means "Nucleo 446RE will need a fair few jumper changes and soldering to permit the firmware to use the required GPIO pins. Refer to the GPIO pins listed below, the solder bridges for these pins should be setup correctly. See Nucleo 446RE documentation UM1724, Section 6.9 on Solder Bridges." I have tried to figure it out without any success from the UM1724 documentation. I am assuming that the jumper changes are on the Neuclo but am confused about the 15 TerminalBlock_bornier-
2_P5.08mm and suspect that the bridges are related to this. Your direction will be highly appreciated **(perhaps an image of the populated board.)

After failing to setup the Toolchain on windows via cygwin, I succeeded in setting it up on a Raspberry Pi 400 and it compiled without any issues. Was wondering if this can be accomplished with CubeMX!

I intend to use the demo driver board to first test this with a toy lathe (Thefirsttool Tz20002mr) that I have had for a while, using Nema 17 motors. And Eventually install on My Sieg C3.

I am located in the Sydney region and have extra PCB's will be happy to send you a couple of sets if you need it considering you are in Australia. My email: [kzaliaust@gmail.com](mailto:kzaliaust@gmail.com). Many thanks for your help

Regards
Kamran Zali

hobbymachinist, 2022-07-23:
> I intend to build it slowly as have beginner level skills (largely in simple Arduino projects.)

Just ask questions on this issue, so it can function as a reference for others. Just setting the expectations and caveats perhaps too late sine you're committed now :) 

- It's a personal project I work on in my spare time - so there's definitely a few rough edges.
- I would normally guide people towards Acorn, Masso, Mach4 or even LinuxCNC - I only started building this because I like using my lathe manually most of the time, ELS is something that helps with repetitive work / tapers etc.
- Backlash compensation is extremely tricky to get right, especially on Z with the way half nuts engage determining the backlash at times. So backlash can potentially vary over time in my observation, needs calibration every single time before you do something.
- I have installed an optical linear scale with 5uM resolution and use this for closed loop feedback instead of backlash compensation.

> I am having a hard time sourcing the FT24C64A EEPROM, CH340C USB UART and the GL850G USB HUB Controller, here in Australia.

I would skip the USART and USB HUB controller, it's only there because I was initially planning to use the HUB for a second daughter board in the future. 

> I will however directly wire PA9 and PA10 to the CN3 connectors marked RXand TX on the Nucleo.

This will work fine, I use this for my second dev kit I use for testing.

> There seem to be many i2c 8 pin 64k EEPROMS on au.rsonline and other suppliers! Any suggestions/alternate will be greatly appreciated.

You should be able to use any 24C64A eeprom in the compatible SO-8 package. They all have the same pinouts and usually have the same read / write control loop / implementation.

>  I have tried to figure it out without any success from the UM1724 documentation. I am assuming that the jumper changes are on the Neuclo but am confused about the 15 TerminalBlock_bornier-2_P5.08mm and suspect that the bridges are related to this.

Can you elaborate ? for each of the GPIO listed in the UM1724 solder bridge section, you need to remove or add the jumper as explained in the user manual to allow the pin to be routed to the external GPIO (disconnected from the default Nucleo function).

Additionally you will also need to
- solder the bridges for the external high frequency (HSE) crystal.
- solder 2 capacitors, I think 18pF

> After failing to setup the Toolchain on windows via cygwin, I succeeded in setting it up on a Raspberry Pi 400 and it compiled without any issues. Was wondering if this can be accomplished with CubeMX!

I use libopencm3, so CubeMX will no help. If you're using Windows 10, you should have better luck with WSL and ARM GCC.

http://cs107e.github.io/guides/install/wsl/

> I intend to use the demo driver board to first test this with a toy lathe (Thefirsttool Tz20002mr) that I have had for a while, using Nema 17 motors. And Eventually install on My Sieg C3.

Yeah go for it, pay attention to polarity, DRV8825 is very picky, fries easy and needs a heat sink.

I have a fair few extra boards around, once the STM32 shortage eases I will most likely respin the board to ditch the Nucleo 64. The only reason I went with that is it was it was impossible to get STM32F446 on tape when I started building this :)

Good luck!

hobbymachinist, 2022-07-23:
and use shielded cables for everything, DC motors or VFD + AC 3ph motor, TFT etc everything is a noise generator, so shield everything :|

kzali, 2022-07-23:
Hi many thanks for a prompt response
> * Pay attention to cable shielding, if possible use ferrite ribbon cable for the TFT (TFTs are very noisy at high speed).
Noted had to do shielded cabling even for a Grbl controller
 
> I would skip the USART and USB HUB controller, it's only there because I was initially planning to use the HUB for a second daughter board in the future.
>Excellent thanks
> 
> You should be able to use any 24C64A eeprom in the compatible SO-8 package. They all have the same pinouts and usually have the same read / write control loop / implementation.

I had meant "There seem to be many i2c 8 pin 64k EEPROMS on au.rsonline and other suppliers!  But  NO 24C64A in SO-8 (seems to be a shortage of these)  So any suggestions?
> 
> > I have tried to figure it out without any success from the UM1724 documentation. I am assuming that the jumper changes are on the Neuclo but am confused about the 15 TerminalBlock_bornier-2_P5.08mm and suspect that the bridges are related to this.
> 
[image]

> Can you elaborate ? for each of the GPIO listed in the UM1724 solder bridge section, you need to remove or add the jumper as explained in the user manual to allow the pin to be routed to the external GPIO (disconnected from the default Nucleo function).
> 
> * solder the bridges for the external high frequency (HSE) crystal.  **Still not clear**
> * solder 2 capacitors, I think 18pF  **That is clear and in your original documents**

> 
> Yeah go for it, pay attention to polarity, DRV8825 is very picky, fries easy and needs a heat sink.
> **I have fried a few in another project need to be careful**
> I have a fair few extra boards around, once the STM32 shortage eases I will most likely respin the board to ditch the Nucleo 64. The only reason I went with that is it was it was impossible to get STM32F446 on tape when I started building this :)
> Great that would be cool 
> Good luck!
Thankyou

deepfryed, 2022-07-23:
> I had meant "There seem to be many i2c 8 pin 64k EEPROMS on au.rsonline and other suppliers! But NO 24C64A in SO-8 (seems to be a shortage of these) So any suggestions?

wow, even eeproms ?! geee

try LCSC https://lcsc.com/products/EEPROM_502.html?keyword=24C64A

> I am assuming that the jumper changes are on the Neuclo 

Yes, all of the jumper changes are on the Nucleo 64 board. No need to to bodge anything on the ELS main board PCB, unless you need to customise something.

You will need to do some point to point wiring for DRO connection if you choose to add that.

hobbymachinist, 2022-07-23:
> I had meant "There seem to be many i2c 8 pin 64k EEPROMS on au.rsonline and other suppliers! But NO 24C64A in SO-8 (seems to be a shortage of these) So any suggestions?

wow, even eeproms ?! geee

try LCSC https://lcsc.com/products/EEPROM_502.html?keyword=24C64A

> I am assuming that the jumper changes are on the Neuclo 

Yes, all of the jumper changes are on the Nucleo 64 board. No need to to bodge anything on the ELS main board PCB, unless you need to customise something.

You will need to do some point to point wiring for DRO connection if you choose to add that.

hobbymachinist, 2022-08-01:
@kzali did you manage sourcing the EEPROM, this might be pin and protocol compatible but half the storage size which may be enough (you need to double check the datasheet) and is available from Rockby in Melbourne. The firmware code needs to be modified if you use this.

https://www.rockby.com.au/SearchresSql_12.cfm?Searchkey=AT24C32N

kzali, 2022-08-01:
> @kzali did you manage sourcing the EEPROM, this might be pin and protocol compatible but half the storage size which may be enough (you need to double check the datasheet) and is available from [...]

@hobbymachinist Thanks for the heads up,  I will be visiting my country of origin and have a micro Shenzhen, in my home town and have sourced all the components there.  At better prices that Aliexpress as well!  In fact will be carrying one set of boards as its very cheap to have them populated by professional services.  Will probably bring some extra components, for future projects.  Have the keyboard with me and the encoder and the pulse wheel expected today.  

Still was only able to setup the toolchain on the raspberry pi 400 while on WSL and a Ubuntu VM I get this error  (Don't know what I am doing wrong!
[image]
So googling away

hobbymachinist, 2022-08-01:
It might be easier to read if you copy paste the error text. The screenshot is hard to read on my phone.

I can see `-I/home/billa/els/libopencm3/include`  is there a libopencm3 directory there ?

> Will probably bring some extra components, for future projects.

excellent, see if you find a cut tape of stm32f446re at good price lol

kzali, 2022-08-01:
billa@billa:~/libopencm3$ cd ~
billa@billa:~$ cd els
billa@billa:~/els$ arm-none-eabi-gcc  --version
arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

billa@billa:~/els$ make
rules.mk:60: Cannot find libopencm3 library in the standard search paths.
rules.mk:61: *** Please specify it through OPENCM3_DIR variable!.  Stop.
billa@billa:~/els$ export OPENCM3_DIR=$PWD/libopencm3
billa@billa:~/els$ make
Using /home/billa/els/libopencm3 path to library
arm-none-eabi-gcc -Os -std=c99 -ggdb3 -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -DGIT_SHA=\"9fbd06c\" -DBUILD_TS=\"2022-08-02T09:30:14\" -Isrc -DVERBOSE=0 -std=gnu99 -mfloat-abi=hard -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Os -DSTM32F446 -DSTM32F4xx -MD -Wall -Wundef -DSTM32F4 -I/home/billa/els/libopencm3/include  -o obj/src/convex_ext_l.o -c src/convex_ext_l.c
src/convex_ext_l.c:9:10: fatal error: libopencm3/cm3/dwt.h: No such file or directory
    9 | #include <libopencm3/cm3/dwt.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [rules.mk:187: obj/src/convex_ext_l.o] Error 1
billa@billa:~/els$

hobbymachinist, 2022-08-02:
@kzali Is there a `/home/billa/els/libopencm3/include/libopencm3` folder ? I think that's what gcc is complaining about

kzali, 2022-08-02:
billa@billa:~$ cd libopencm3
billa@billa:~/libopencm3$ ls
COPYING.GPL3  COPYING.LGPL3  HACKING  HACKING_COMMON_DOC  Makefile  README.md  doc  els  include  ld  lib  locm3.sublime-project  mk  scripts  tests
billa@billa:~/libopencm3$ cd include
billa@billa:~/libopencm3/include$ ls
libopencm3  libopencmsis

yes the folders are there

hobbymachinist, 2022-08-02:
heh, `-I/home/billa/els/libopencm3/include` so that's wrong

it should be /home/billa/libopencm3, check your OPENCM3_DIR env variable

kzali, 2022-08-02:
Thanks will check that and give it a shot
Alas my terrible Linux skills.  Appreciate your help, I guess that's how
one learns!

kzali, 2022-08-02:
> heh, `-I/home/billa/els/libopencm3/include` so that's wrong
> 
> it should be /home/billa/libopencm3, check your OPENCM3_DIR env variable

@hobbymachinist  Thanks cleaned the env variable's and export OPENCM3_DIR=$PWD/libopencm3
vola it worked like magic!!  Thanks to you learnt a lot today.
Cheers

closed 2022-08-20 by hobbymachinist