hobbymachinist/els #6: Are there compatible dev boards?

gerritv, 2022-10-11:
STM32's are rare at present, when they become available again are other version ok to use? I realized pin assignements might change but if it is STM32 based will that be ok? I think LPC1768 support in libopencm3 is not complete else I would ask about that as well.

hobbymachinist, 2022-10-11:
I initially wrote this to be compatible with both 103CB, 441RE and 446RE but ended up ditching the first 2 due to processing and gpio requirements. So at this stage unfortunately no and it looks like they're sold out everywhere until mid 2023, sorry. I am planning to redesign this sometime later this year or early next, I can look at other options then if the shortage persists.

hobbymachinist, 2022-10-11:
@gerritv if you're comfortable designing and building your own pcb, LCSC have some STM32F446RET6 in stock https://lcsc.com/search?q=stm32f446ret6

closed 2022-10-11 by hobbymachinist

gerritv, 2022-10-12:
Thank you, I think I will work at being patient, some sites say November for the Nucleo. 
(The RELS I am using is passable for now but look forward to moving to something decent soon. )

hobbymachinist, 2022-10-12:
@gerritv I wouldn't consider this anywhere close to implementations like ELSv4 Pro from Rocketronics. As noted in the documentation, this is very much experimental and fairly complex build to add to that. If you're looking to get a good ELS replacement, I'd strongly recommend the Rocketronics ELSv4 basic or pro. The BOM cost for this project is pretty close to the ELSv4 basic pricing anyways, and that has professional support and nice features such as tool radius compensation.

If you're like me and like tinkering soldier on :)

gerritv, 2022-10-12:
I don't mind tinkering. Even with the Euro at a better rate of conversion than usual, it is too much $ all at once.
Presently I use RELS7e2. It works but there are some issues with large thread pitches, and also with small ones. The code is not pretty. The hard work IMO is adding the X stepper drive, and with both steppers in place, swapping controllers is not a big issue.

Would one of the M4 based Nucleo's work?

hobbymachinist, 2022-10-12:
M4 based Nucleos ? You mean https://www.mikroe.com/mini-stm32f4 ? It may work, I'll have to go through the datasheet for STM32F415RG but STM32s are generally pin compatible within one family but the peripheral / AF config might be slightly different.

gerritv, 2022-10-12:
I was thinking along these lines: https://www.digikey.ca/en/products/detail/stmicroelectronics/NUCLEO-L452RE-P/7380300

hobbymachinist, 2022-10-12:
L452s are low power, so run much slower. It's going to struggle handling the TFT and other real time tasks at the same time. I could get it barely running without jitters on the 411RE that runs at 125MHz. I haven't spent a lot of time optimising the control loop, but the LCD refreshes are expensive even with DMA.

gerritv, 2022-10-12:
Can you confirm that this is The One please? yesterday there were 0, now.... https://www.mouser.ca/ProductDetail/STMicroelectronics/NUCLEO-F446RE

hobbymachinist, 2022-10-12:
STM32F429ZI, yeah that'd work. It's Nucelo-144 though, so you'd have to make your own break out board in addition to customising the gpio config to suit it but doable.

hobbymachinist, 2022-10-12:
mouser Australia has stock, yes it's the part number you have https://au.mouser.com/ProductDetail/511-NUCLEO-F446RE

gerritv, 2022-10-12:
it is for evaluatoin of STM32F446RET6 so I will order 2 .
thank you for your patience and help :-)

hobbymachinist, 2022-10-12:
Haha, I'll order an extra board for myself. Thanks for scouring the interwebs :laughing:

gerritv, 2022-10-14:
Nucleo boards arrived, waiting on @kzali for mailing costs for a set of breakout boards. Would love to get a second set as I am buiding one for myself, one for a friend.

AndreyFly007, 2022-10-23:
> but the LCD refreshes are expensive even with DMA.

display less frequently
Clockwise / Counter clockwise direction of spindle and RPM. Chuck icon is shown when RPM = 0

remove from screen:
Angular position of the spindle in degrees. This can be used for any indexing functions.
Pulse counter, this is a function of the encoder PPR and wraps around. Threading operations synchronise to this value.

Another option is to install a cheap processor for displaying. Send the necessary data with ELS, and let the cheap processor redraw the screen