hobbymachinist/els #7: Using Platformio and Kicad 6.0

gerritv, 2022-10-13:
I forked the repo and have it compiled using Platformio as a build system. (I use VSCode and Pio for everything)

Are you interested in a pull request?
https://github.com/gerritv/HM-els

KiCAD wanted to 'rescue' some outlines.
Platformio needed its .ini file
A few code changes due to constants being renamed in libopencm3 for the f4. e.g. in clock.c, it is now RCC_CFGR_HPRE_DIV_NONE, previously it was RCC_CFGR_HPRE_NODIV

hobbymachinist, 2022-10-13:
> I forked the repo and have it compiled using Platformio as a build system. (I use VSCode and Pio for everything)

I would recommend creating a separate repo with els vendored and putting your local IDE / build config in that repo. This keeps the upstream clean and allows you to customise your build / dev environment.

> KiCAD wanted to 'rescue' some outlines.

Yes, I'm in the middle of migrating the project to Kicad 6. I'll push some updates soon.

> A few code changes due to constants being renamed in libopencm3 for the f4. e.g. in clock.c, it is now > RCC_CFGR_HPRE_DIV_NONE, previously it was RCC_CFGR_HPRE_NODIV

I'll look into that as well, thanks. I'll have to test with new version of libopencm3, I might have to consider pinning libopencm3 versions to avoid build issues.

hobbymachinist, 2022-10-13:
I've pushed the initial Kicad 6.0 migration / changes over. I haven't updated the add on daughter boards, but that should be straightforward.

https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/stm32/f4/rcc.h#L228  that's still RCC_CFGR_HPRE_DIV_NONE 

are you using a different version / tag ?

hobbymachinist, 2022-10-13:
@gerritv Why have you removed the symlinks ?

https://github.com/hobbymachinist/els/compare/main...gerritv:HM-els:main#diff-a37ff317ec9d68c5ca74e2471f6b7e428bf47a510c81a34918cd7cbe19e5008eR1

closed 2022-10-13 by hobbymachinist

gerritv, 2022-10-13:
I changed the clock.c file from RCC_CFGR_HPRE_NODIV to RCC_CFGR_HPRE_DIV_NONE. ..._DIV2 becomes DIV_2 etc. I'm assumin gthe libopencm3 is newest in Platformio.

gerritv, 2022-10-13:
`"I would recommend creating a separate repo with els vendored and putting your local IDE / build config in that repo. This keeps the upstream clean and allows you to customise your build / dev environment.`
I'll figure out how to do that over the next day or so, tku.

hobbymachinist, 2022-10-13:
libopencm3 link above is from their github repo, it's the latest. I suspect the version in platformio is old (at least by 2yrs). I recommend using the latest version.

gerritv, 2022-10-13:
> @gerritv Why have you removed the symlinks ?
> 
> [main...gerritv:HM-els:main#diff-a37ff317ec9d68c5ca74e2471f6b7e428bf47a510c81a34918cd7cbe19e5008eR1](https://github.com/hobbymachinist/els/compare/main...gerritv:HM-els:main#diff-a37ff317ec9d68c5ca74e2471f6b7e428bf47a510c81a34918cd7cbe19e5008eR1)

I was getting dup references on the link, but  may have solved that another way. Also  I have never come across that construct before  so 'solved it' :-)

hobbymachinist, 2022-10-13:
you need to exclude the platform files, see the Makefile for guidance

https://github.com/hobbymachinist/els/blob/main/Makefile#L5-L6

gerritv, 2022-10-13:
> libopencm3 link above is from their github repo, it's the latest. I suspect the version in platformio is old (at least by 2yrs). I recommend using the latest version.

Platformio's version is same as what you listed, the code in clock.c then was old? clock.c was using NODIV.

gerritv, 2022-10-13:
> you need to exclude the platform files, see the Makefile for guidance
> 
> https://github.com/hobbymachinist/els/blob/main/Makefile#L5-L6

I excluded them in platformio.ini, but tft.c gives me this error:

src\tft\tft.c:1:9: error: expected '=', ',', ';', 'asm' or '__attribute__' before '/' token
 platform/tft_opencm3_stm32fx.c"

I'll dig into that tomorrow, its late up here.

hobbymachinist, 2022-10-13:
> https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/stm32/f4/rcc.h#L228 that's still RCC_CFGR_HPRE_DIV_NONE

I meant to say that's still RCC_CFGR_HPRE_NODIV in the latest libopencm3 and the constant in src/clock.c is correct.  'night.

gerritv, 2022-10-13:
A total lack of versioning of libopencm3 made this extra challenging, I guess ppl forgot how to incorement file version #'s :-( I found a solution though (for the benefit of others who run into this issue): https://community.platformio.org/t/compiler-doesnt-find-function-definition-libopencm3/25795/5 . This approach would make a configuration manager freak out but it works. And Platformio is behind in updating libraries. (Given where they are in the world it is understandable.)

So now back to the symlink in tft.c. Can you give me a lead on where to read about this construct? Searching the web for anything symlink only returns stuff about file systems.

hobbymachinist, 2022-10-13:
> So now back to the symlink in tft.c. Can you give me a lead on where to read about this construct? Searching the web for anything symlink only returns stuff about file systems.

I've pushed some changes to remove the symlinks, they're essentially symbolic pointers in the file system to actual files. You should be able to compile them as is now.

hobbymachinist, 2022-10-13:
it's a bit late here, I'll check back tomorrow.

gerritv, 2022-10-13:
Thank you for your help and patience.
Ok, all solved. Tku for removing the symlink, not common on windows. I thought it was some magic need C standard thing :-)

By adding a platfrom.ini file and a python script to get output from git rev-parse, I now have a clean build.

I can do a pull request for you to add those 2 files, or I can keep it local to my PC. 

For others:
platformio.ini:
[env:nucleo_f446re]
platform_packages = framework-libopencm3 @ https://github.com/larmicfre/libopencm3.git
platform = ststm32
framework = libopencm3
board = nucleo_f446re
build_flags = 
    !python git_rev_macro.py
    -Wno-unused-variable
    -Wno-unused-function 

git_rev_macro.py:
import subprocess

revision = (
    subprocess.check_output(["git", "rev-parse", "HEAD"])
    .strip()
    .decode("utf-8")
)
print("'-DGIT_SHA=\"%s\"'"%revision)

hobbymachinist, 2022-10-13:
If I add it to upstream here, I'll have to maintain it and make sure I don't commit breaking changes. Since I don't use platformio or vscode, it might be something better suited for you to maintain locally.