hobbymachinist/els #10: LCD ?

beyma1, 2023-05-16:
Brak możliwości zakupu wyświetlacza ILI9481. Zakupiłem ILI9486 oraz ILI9488, na żadnym z nich nic się nie wyświetla

hobbymachinist, 2023-05-17:
Sorry, I don't understand Polish but I assume you're asking if it will work with ILI9486 or ILI9488. It will not, you need to write a new software driver for those. I don't have those LCDs to test so I can't help you at this time, sorry.

closed 2023-05-17 by hobbymachinist

hobbymachinist, 2023-05-19:
I purchased a ILI9488 display, I'll have to wait a month or so for it to arrive and I can take a look at adding support for it once I have it.

beyma1, 2023-05-20:
Dzieki.... fajnie by bylo gdyby sie udalo przerobic kod by els pracowal pod ili9488. W oryginalnym projekcie wykorzystany jest ili9481 ktory praktycznie jest juz nie dostępny. Ostatnio udalo mi sie znalezc u chinczyka ale wersja z dotykiem. Zamowilem go i tez przyjdzie za jakis miesiąc hehe. Ciekawe czy ten wyswietlacz to naprawdę przyjdzie ili9481 albo jakis fake.... czas pokaże. pozdrawiam

Thanks .... it would be nice if you could change the code so that els would work under ili9488. The original project uses ili9481, which is practically no longer available. Recently, I managed to find a Chinese version, but with a touch. I ordered it and it will arrive in about a month hehe. I wonder if this display will really come ili9481 or some fake .... time will tell. Regards

beyma1, 2023-05-21:
"Refer to the GPIO pins listed below, the solder bridges for these pins should be setup correctly. See Nucleo 446RE documentation UM1724, Section 6.9 on Solder Bridges."
Could you please explain which jumpers are you talking about? Do these jumpers also affect the operation of the display?

hobbymachinist, 2023-05-21:
https://github.com/hobbymachinist/els/blob/main/docs/els.md#solder-bridge-changes-and-modifications-needed-to-nucleo64-446re 

Hope that helps

beyma1, 2023-05-21:
I made all the jumpers as you described .... However, I missed one SB51 jumper. After its disassembly, ELS works on both displays ... ILI9486 and ILI9488

beyma1, 2023-05-21:
Although the colors are different than yours with the ILI9481 display, but the fact WORKS :)

hobbymachinist, 2023-05-21:
> However, I missed one SB51 jumper. 

@beyma1 You mean SB21 ? SB51 is for PC0, which is unused. Glad you got it working, the colors are hardcoded to ILI9481, I'll have a look at updating them for ILI9488 when I get my LCD.  If you can share a video or some photos once you got the ELS working, I'd appreciate it :)

beyma1, 2023-05-23:
Of course, I meant the SB21 jumper (my spelling error .... Already corrected)

hobbymachinist, 2023-05-23:
You're nearly there, hope you get it working soon and are happy with it :)

beyma1, 2023-05-25:
I'm already satisfied... The mere fact that the lcd lights up makes me happy. Can the keyboard be replaced with something else? I don't like to spend so much money on a dedicated keyboard that was used in the project. Maybe there are alternatives?

hobbymachinist, 2023-05-25:
You can get a cheap PS/2  keypad and use that. +5V, GND, Data (PC5) and Clock (PC4) from the keypad should go to the keypad connector

[image]

hobbymachinist, 2023-05-26:
yeah that's not too hard. PS/2 19-20 key plastic keypads are very cheap $15-20, if you want to save yourself the trouble. You can remove the standard enclosure and 3D print a mounting plate for it. Only change needed to firmware would be changes to the scan codes in constants.h

hobbymachinist, 2023-05-30:
Did you have to change any initialisation sequence / code for ILI9488 and ILI9486 ? I received a ILI9488 today, I don't think they work out of the box.

beyma1, 2023-05-30:
The displays worked without any modifications. Today I received the display and as it is with the Chinese instead of ILI9481 I received ILI9488 ehh. This display works too. So theoretically all these displays are interchangeable, only the colors are different (as you can see in the photos I posted earlier)

hobbymachinist, 2023-05-30:
haha,  I got 2 that are supposed to be ILI9488. They don't work at all, I'm not sure I want to try debugging this :)

I'll have a look at the datasheet when I find time and investigate. Re. colours, the color byte order might be different that what the ILI9481 driver uses (BGR). You can try swapping them.

hobbymachinist, 2023-05-31:
@beyma1 do you have a link to where you purchased ILI9488 from ? thanks

hobbymachinist, 2023-05-31:
@beyma1 I got ILI9488 working, looks like the command set is entirely compatible, but I need some time to review the ILITEK datasheet to confirm that the initialisation sequence is not going to affect the display in the long term.

https://github.com/hobbymachinist/els/blob/main/Makefile#L25   comment out the default ILI9481, uncomment ILI9488 and give it a try. Colors look fine to me, it's mostly a controller timing issue. ILI9488 seems slower, so needs additional delay to process signal transitions.

hobbymachinist, 2023-06-03:
updated ILI9488 initialisation, please use the latest driver code.

beyma1, 2023-06-15:
I checked and the new codes work. And the funniest thing is that I have two ili9486 and two ili9488 displays (from two different suppliers) and only one of them works .... The Chinese man sold me some broken displays. Well, it's hard. In some time I will try to handle the keyboard on arduino nano, the board is already made, just solder and program it (if I find time). Recently, I've also been thinking about replacing this display with a DWIN with a touch panel .... It would also be nice to implement this driver ... But it will probably stay in the dream phase

hobbymachinist, 2023-06-15:
This is where I got the original ILI9481 and the ILI9488s from https://www.aliexpress.com/item/1005001288341298.html

Seems reliable.

Touchpanels are problematic in the machine shop, especially capacitive touch (due to electrical noise), oil and dirt in the fingers also pose an issue. I wouldn't recommend.