hobbymachinist/els #12: Taper cutting accuracy

klandrith, 2024-08-31:
Hello,
I have watched all your videos and read all of the documentation on this ELS. In one of your videos you mentioned that the taper cutting accuracy was less than ideal, but you were hopeful that closed loop linear scale feedback would alleviate this. 

I haven't found any mention of how it ended up working out. I am considering outfitting my Harrison M300 lathe that is missing it's factory taper attachment with this ELS to add that functionality as well as other goodies. Any help would be appreciated, and thank you for the cool project.

hobbymachinist, 2024-08-31:
tl;dr without good quality ballscrew conversion, it's not precise but I've got acceptable results with conservative cuts and then polishing / grinding off high spots to finish. 

If you're cutting tapers with the top / compound slide, then the accuracy of the taper is only dependent on the linear accuracy of the dovetail and gibs which would be very good. Even with closed loop setup, ELS only compensates for backlash at the start of the cut at the moment.  So the accuracy is now a function of the the accuracy of the following:

1. Z leadscrew
2. Z halfnuts
3. X leadscrew
4. X halfnuts
5. cross slide dovetail & gibs

I've tried continuous compensation with varying results, none satisfactory because the DRO scales themselves have errors of up to +/- 5um. If I need a taper turned with high precision, I still bring out my compound slide but I'm planning to make a custom taper cutting attachment with linear rails in the future.

hobbymachinist, 2024-08-31:
>  I am considering outfitting my Harrison M300 lathe that is missing it's factory taper attachment with this ELS to add that functionality as well as other goodies.

If you're comfortable tinkering with electronics and firmware go for it, otherwise something like https://www.rocketronics.de/shop/en/controls/lathecontroller/ would be a better option.

closed 2024-08-31 by hobbymachinist